jniebuhr/gaggimate #21: Allow buttons to be used

jniebuhr, 2024-11-04:
The survey showed that people want to be able to still use their physical buttons on the device.
I suggest the following actions:

- Standby Mode and Brew / Steam Button switched on -> Switches to Brew/Steam screen and pre-heats
- Brew Mode and Brew button switched on -> Starts brew process
- Brew Mode, brew process running and brew button switched off -> Stops brew process
- Steam Mode and Steam button switched on -> Start steam process
- Steam Mode, Steam running and button switched off -> Stops steam process

Toggling on the buttons when the process is already running or toggling them off when nothing is running will have no effect

mbicknese, 2025-01-03:
It might be nice to have the brew group disperse hot water when the system is in standby mode and the brew button is toggled. Basically, the default behaviour of the machine. 

This would allow users to heat up the brew group and portafilter. 

Potentially also great for back flushing.

jniebuhr, 2025-02-17:
Perfect thanks @boraca, do you want to contribute a page to https://github.com/jniebuhr/gaggimate-docs ?

hardikdabhi, 2025-02-18:
US models do not have spring, better to keep functionality as initial?

PR #139 (2025-03-15): feat: Add button control
Fixes #21
closed 2025-03-15 by jniebuhr