jniebuhr/gaggimate #38: PCB Next Revision

jniebuhr, 2024-12-06:
This should be a collection for what should be changed on the PCB.

jniebuhr, 2024-12-06:
Remove Vias from Pads of R3, C10, R14, R15, U3

jniebuhr, 2024-12-06:
Replace JST XH with JST PH (possibly with dual purpose pads SMD/PTH)

jniebuhr, 2024-12-06:
Add bigger capacitor for 5V

jniebuhr, 2024-12-06:
Add Fuse and Inductir for 230V

jniebuhr, 2024-12-06:
Add snubber circuit to 230V (unless going for SSR instead of Relay)

oyvkar, 2024-12-06:
> Replace JST XH with JST PH (possibly with dual purpose pads SMD/PTH)

Remember to anchor SMD pads with vias at ends/edges in order to prevent tearing off SMD connector pads.

jniebuhr, 2024-12-06:
Optionally replace relays with SSRs

jniebuhr, 2024-12-06:
Add pressure sensor, dimmer circuit (separate version + expansion)

jniebuhr, 2024-12-06:
Replace 2 separate connectors for mains with single one

jniebuhr, 2024-12-06:
See how grinder/alt port is used and keep/remove it for future versions

jniebuhr, 2024-12-06:
People asked for smaller form factor. This probably means more difficult assembly, do a poll on this

jniebuhr, 2025-02-04:
Use a different color connector for SSR to make it less likely for people to connect the SSR to the screen

JamesCarron, 2025-05-03:
Hi are the pro pcb design files available? The PCB files I have open (pcb-gen2 branch) look different to the board shown on the shop page.

closed 2026-10-05 by jniebuhr