jniebuhr/gaggimate #56: Add “cleaning” option

aidenlikescoffee, 2025-01-17:
Add option for cleaning to be able to backflush

jniebuhr, 2025-01-17:
This will be easier to add after #53 so should be done after it's merged

jniebuhr, 2025-01-20:
MR is now merged, someone could implement this 👍

sawtell, 2025-03-17:
Not sure if this is the right place or a separate ticket would be best, but a descaling program would be great too.

jniebuhr, 2025-04-10:
I could just add both as preset profiles, I guess that should be enough? What do you think @sawtell @aidenlikescoffee

sawtell, 2025-04-10:
@jniebuhr That's a nice solution, but would that mean the feature is only available to those with the Pro board and pressure profiles?

jniebuhr, 2025-04-10:
@sawtell no, everyone gets profiles, just only GaggiMate Pro gets Pressure Profiling.
For either installation you can create basic or pro Profiles

timothe, 2025-10-30:
imho the profile route is messy. It doesn't show the user what to do and when, plus there is a timeout issue I pointed out #473