jniebuhr/gaggimate #108: [Compatibility Request] Machine Name - Model

krskt, 2025-02-18:
### Producer of the machine

Lelit

### Machine model

Anna PL 41 TEM

### Does the machine have a 3-way valve?

Yes

### What method of heating does your machine use?

Thermoblock

### Additional context

Thanks!

jniebuhr, 2025-02-18:
[image]

jniebuhr, 2025-02-18:
Looks like everything would be there. I didn't see an electrical diagram but with these machines there's often a sensor to keep that boiler filled I think. Would need to keep that in place and working.

krskt, 2025-02-18:
Found diagram of a similair modell 

[image]

krskt, 2025-02-18:
Here. I dont have really good idea of wirings. I was even considering selling the machine and getting the Gaggia and put your mod :) What would you recommend?

[image]

jniebuhr, 2025-02-18:
do I understand it correctly that the steam switch overrides the PID setting?

jniebuhr, 2025-02-18:
the last one you posted seems to be from a machine without PID

krskt, 2025-02-18:
> do I understand it correctly that the steam switch overrides the PID setting?

If it means when i switch on the steam the Temperature display goes to 135.

jniebuhr, 2025-02-18:
but yeah, shouldn't be a problem to install it. just keep in mind that there'll probably not be a guide for it

krskt, 2025-02-18:
> the last one you posted seems to be from a machine without PID

Ah you are right!

jniebuhr, 2025-02-18:
> Found diagram of a similair modell
> 
> [image]

on this one the IV (Steam Switch) bypasses the line through the SSR.

krskt, 2025-02-18:
Oh i cant find the right diagram for that Machine... And I am scared to do it without a proper guide. Thank you anyway:)
If i go for the Gaggia. Which model should i look for? I heard the Evos are not recommendable. ?

jniebuhr, 2025-02-18:
you do not want the teflon coated boilers. from where you're coming from maybe just go for an E24 machine. I can also recommend a Rancilio Silvia.

jniebuhr, 2025-02-18:
[image]

jniebuhr, 2025-02-18:
this looks like it's very similar to the diagram you posted.

closed 2025-02-18 by jniebuhr