jniebuhr/gaggimate #109: [Compatibility Request] Machine Name - Model

jjvangalen2001, 2025-02-18:
### Producer of the machine

Bezzera 

### Machine model

BZ10

### Does the machine have a 3-way valve?

No

### What method of heating does your machine use?

Heat Exchanger

### Additional context

one guy on github did something but not sure if he also added pid controll. https://github.com/ggramlich26/BZ10_upgrade. On the repair manual is says 2 way solenoid valve not sure if you can change it to a 3 way .but on page 25 it does use a 3 way solenoid valve.

[BZ10 9942047.06 GB ed.09.21 rev00.pdf](https://github.com/user-attachments/files/18851599/BZ10.9942047.06.GB.ed.09.21.rev00.pdf)

jniebuhr, 2025-02-18:
I checked the electrical diagram of this and as it currently stands you'd need to combine some parts of the original circuitry with GaggiMate controlled functions.

jniebuhr, 2025-02-18:
[image]

jniebuhr, 2025-02-18:
L/A on this picture appears to be the tank fill solenoid, this function would need to still be controlled by the machine

jjvangalen2001, 2025-02-18:
thanks for checking

closed 2025-11-05 by jniebuhr