jniebuhr/gaggimate #112: [Compatibility Request] Machine Name - Model

bernardoregan94, 2025-02-19:
### Producer of the machine

Gaggia 

### Machine model

New baby 06 twin 7464 

### Revision (if applicable)

2009

### Does the machine have a 3-way valve?

Yes

### What method of heating does your machine use?

Single boiler

### Additional context

It has a single boiler for brewing and a thermo block for steam I believe.

bernardoregan94, 2025-02-19:
[image]
[image]

jniebuhr, 2025-02-19:
This one might be a bit harder since it has components that GaggiMate currently has no support for. With a dual boiler extension coming there should be a way to make this work though. I'll leave it open for now.

bernardoregan94, 2025-02-19:
> This one might be a bit harder since it has components that GaggiMate currently has no support for. With a dual boiler extension coming there should be a way to make this work though. I'll leave it open for now.

Thanks mate