jniebuhr/gaggimate #117: [Bug] Scale reconnect logic

DevNullGamer, 2025-02-25:
### Brief summary of the bug

Scale doesnt alway reconnect without a restart of the display

### Describe the issue in greater detail

When coming back to GaggiMate that has been in standby between shot, i experience that i have to restart the display to get the scale to reconnect.
It does not reconnect by itself in most cases i have experienced, but have just used a work around by restarting the screen, and then the scale connects within seconds.

### Steps to reproduce

1. Start and connect scale to GaggiMate
2. Turn off scale and wait a little while (dont know the exact amount of time)
3. Turn on the scale again and wait to see if it reconnects
   A. If it reconnects, try again with longer period of time between turning off the scale and on again
   B. If it doesnt connect, you have reproduced the issue

### Are you able to reproduce the issue?

Yes

### Expected behavior

That the scale and screen reconnects automatically.

### Has this function/fetaure worked before with your GaggiMate?

No

### Relevant logs or error messages

```shell
No logs available currently.
```

### What machine are you experiencing this issue on?

Gaggia Classic Pro 2018
Eureka Precisa bluetooth scale.

### Additional context

N/A

jay-jey, 2025-03-14:
can confirm the Bug with my bluetooth scale. Same model

jniebuhr, 2025-03-14:
Did you update to the newest version? That should've fixed it

jay-jey, 2025-03-14:
Yes I have 1.3.3 and it is still there for me.

jniebuhr, 2025-03-19:
Should be fixed by #158

closed 2025-03-26 by jniebuhr