jniebuhr/gaggimate #118: [Compatibility Request] Breville/Sage Dual Boiler

norcrazy88, 2025-02-26:
### Producer of the machine

Breville / Sage

### Machine model

Dual Boiler

### Revision (if applicable)

BES920

### Does the machine have a 3-way valve?

Yes

### What method of heating does your machine use?

Dual boiler

### Additional context

Considering buying the Breville/Sage Dual Boiler . With all the functions of this machine from factory, would it still be a good idea to mod this?
Attached pdf with blueprint/parts list.
Cannot find a official electrical diagram, but there seems to be a legit drawing in forum thread here https://www.home-barista.com/repairs/breville-bes920-dead-after-o-ring-change-t88957.html

Breville/Sage brand is used for different markets (US/EU etc)
Other revision of this machine is BES900, but I have not looked into what changes were made. BES900XL/BES920XL model numbers seems to be no different from the non XL versions (not fully confirmed).

This seems to confirm a 3-way valve as model names states fitting https://www.coffeesparesdirect.co.uk/SP0001624-SAGE-TOUCH-ORACLE-3-WAY-SOLENOID

[56615_74db63937352e9eb5cc796d2849b4c60.pdf](https://github.com/user-attachments/files/18975530/56615_74db63937352e9eb5cc796d2849b4c60.pdf)

thrOo, 2025-05-15:
Also interested in this. I have another version of it (Gastroback 42636 Advanced Control Dual Boiler).
That is an equivalent on the Breville Dualboiler Version1 (without the boiler water outlets).
I won't have time to tinker around until late this year, but I am open to test if applicable.

Ftrlvb, 2025-06-04:
will the separate steam boiler be a problem?

atomic004, 2025-06-11:
I would love to install this in my Sage Dual boiler. I have a very recent one (nov 2024).
I don't mind testing or experimenting with some stuff if you want.

kojinism, 2025-06-21:
Hoping for good news soon. :)

markfallu, 2025-12-09:
I am quite interested in this.  It seems to me there are 2 pathways.  1) Change the existing Breville/Sage dual boiler sensors  (NTC thermistors) to k type thermocouples with form factors that compatible with those intended for the NTCs. (Essentially a “hardware” approach.)   2) Leverage the SPI bus and implement and external ADS1220 board that can implement SPI based temperature sensing for the existing NTCs (or RTDs, like a pt 100) - the “software approach”

I am quite interested in option 2 - as it would mean that a wide variety of machines could be made compatible.  It would however involve some changes on the software side.  

Right now there is a class TemperatureSensor that uses class Max31855Thermocouple : public TemperatureSensor

If we added a new class Ads1220Thermistor : public TemperatureSensor, then we could add a runtime configuration to switch between sensor types.

From the PCB schematic it looks like all the required pins are available so an external ADS1220 board could be wired up in a minimally invasive way.  https://www.olimex.com/Products/Breadboarding/BB-ADS1220/open-source-hardware

These are 4 channel - so single board could handle brew boiler, brew head, and steam boiler (with one spare).

There would be a bit of coding required to get equivalent functionality for Rob Tillaart’s MAX31855 Arduino library, but it is doable.

I will look a bit more closely at the schematic and code and report back.

Obscurotes, 2026-03-11:
When BES920 support lands, do you expect the install to be mostly harness/connector-based, or should users plan for soldering?
Also, any chance of supporting a small display that could replace the stock pressure gauge, instead of using a larger external screen?

alex-2000-ac-ghetto, 2026-09-19:
Yes please! Triple thumbs up for the Dual Boiler !! Keen as. cheers