jniebuhr/gaggimate #119: [Compatibility Request] Gaggia - GCP Gaggiuino V3 (or lower)

AnomalRoil, 2025-02-26:
### Producer of the machine

Gaggia

### Machine model

Classic Pro modded with legacy Gaggiuino

### Does the machine have a 3-way valve?

Yes

### What method of heating does your machine use?

Single boiler

### Additional context

It seems the sensors (thinking of mostly the temperature and pressure ones) the Gaggimate relies on are similar to the one the legacy Gaggiuino project relies on, and logically the rest of the controls are the same since the underlying machine is the same.

It would be great to have an upgrade path from the legacy (open-source) Gaggiuino project to the Gaggimate, along with an upgrade guide, for the people who prefers to stick with open-source projects.

jniebuhr, 2025-02-27:
The wiring and connecting components are similar, you can drop in the GaggiMate PCB and Screen. I'm not sure I should make a conversion guide, feels a bit like a declaration of war

AnomalRoil, 2025-03-07:
I see, any risks of you offering an "upgrade kit" then, with just the screen and PCB?

banoz, 2026-01-16:
@AnomalRoil what did you end up with?