jniebuhr/gaggimate #124: [Compatibility Request] Machine Name - Model

alex1ard, 2025-03-06:
### Producer of the machine

Bugatti casa

### Machine model

Diva

### Does the machine have a 3-way valve?

Yes

### What method of heating does your machine use?

Single boiler

### Additional context

The internals are very similar to a gaggia, the boiler is way smaller and only has one heating coil.

alex1ard, 2025-03-06:
[image]
[image]
[image]
[image]
[image]

alex1ard, 2025-03-06:
[image]

jniebuhr, 2025-03-06:
That's a very fancy looking espresso machine but yeah, looks like it's compatible. The boiler looks more like a Rancilio Silvia v1 boiler