jniebuhr/gaggimate #126: [Compatibility Request] Machine Name - Model

jcrisp16, 2025-03-07:
### Producer of the machine

Profitec

### Machine model

Pro 300

### Does the machine have a 3-way valve?

Yes

### What method of heating does your machine use?

Dual boiler

### Additional context

Would absolutely love to know if there's a way to add pre infusion to this machine via what you're building. Thanks!

jniebuhr, 2025-03-12:
It seems like this machine has a completely separate pump for the steam part? In that case I'd recommend to install gaggimate entirely on the brew side and leave the steam side as is.

jcrisp16, 2025-03-12:
@jniebuhr I'm pretty sure it only has one pump.

https://www.wholelattelove.com/products/profitec-pro-300-part-diagram-pro300

jniebuhr, 2025-03-12:
Yes sorry, you're correct. I guess it fills the steam boiler over the OPV? It looks like it's using barb style connectors so you'd choose that option in the shop. I'm not entirely sure which kind of female/male connector it uses everywhere so you might want to stock up on some extra wires/quick connect connectors.

jcrisp16, 2025-03-12:
Ok great and then I assume M3 K-type?

jniebuhr, 2025-04-10:
I'd think M4 if you want to remove something else. The main temperature sensor for the PID seems to be one that measures water temperature with a G1/8 thread

closed 2025-11-05 by jniebuhr