jniebuhr/gaggimate #131: [Compatibility Request] Gaggia - Classic Coffee (2002)

github-fabio, 2025-03-10:
### Producer of the machine

Gaggia

### Machine model

Classic coffee

### Revision (if applicable)

2002

### Does the machine have a 3-way valve?

Yes

### What method of heating does your machine use?

Single boiler

### Additional context

Hey,
I’ve got an old Gaggia Classic Coffee from around 2002. Is it still compatible with GaggiMate, or have there been any significant hardware changes over the years that might make my machine incompatible?

The picture of the inside isnt mine. I just grabbed one off an old redit thread real quick.

[image]
[image]

smury, 2025-03-12:
I think with high confidence that it is compatible. I have essentially the same model - a Gaggia Classic from 2012. Looks identical apart from the 'coffee' in the name and the 2 side screws. These models have the advantage of an adjustable OPV. What's the model number?

jniebuhr, 2025-03-12:
Yeah, there's already been installs on this machine. It's one of the main supported ones.

closed 2025-03-12 by github-fabio