jniebuhr/gaggimate #132: [Compatibility Request] Saeco - Aroma (SIN 015X)

jazzman15, 2025-03-11:
### Producer of the machine

Saeco

### Machine model

Aroma [SIN 015X]

### Revision (if applicable)

2000ish

### Does the machine have a 3-way valve?

No

### What method of heating does your machine use?

Single boiler

### Additional context

The boiler appears to have an M4 thread for the thermostat, so an M4 K-Type thermocouple should be compatible.
The ULKA EX5 vibratory pump is used in this machine.

Will steaming be affected? 

Looking forward to do the build!

[image]

jniebuhr, 2025-03-12:
You can install it on this machine but some features that use a 3WV specifically wouldn't be available

37, 2025-06-03:
@jazzman15 how did you go with the installation? I've just saved an old Saeco Aroma from being thrown out and am thinking of doing the same mod. I'd love to contribute to add support if need be. 

@jniebuhr thanks for the confirmation mate. Out of curiosity what do you look for to assess compatibility and what further modifications to this machine would be needed to fully leverage the other features of the gaggiamate? I'm a machinist and replacing the boiler, hardware mods etc. aren't a concern.

jazzman15, 2025-06-04:
@37 Hey! Thanks you for reaching out. I haven't started the mod yet, as I feel a bit hesitant about upgrading this particular machine. It almost seems like trying to put lipstick on a pig! I'm also concerned about whether there's enough space to accommodate the PCB, which might pose a challenge.

If you're going ahead with the modification or have any thoughts on how to tackle these issues, your contributions would be greatly appreciated!

closed 2025-06-04 by jazzman15