jniebuhr/gaggimate #136: [Compatibility Request] Gaggia Classic - RI9403/18

dlking12, 2025-03-12:
### Producer of the machine

Gaggia

### Machine model

Classic RI9403/18 

### Revision (if applicable)

2015

### Does the machine have a 3-way valve?

No

### What method of heating does your machine use?

Single boiler

jniebuhr, 2025-03-12:
You can install it on this machine but some features that use the 3WV would not be supported.

dlking12, 2025-03-12:
Thank you. Would this be features such as pressure control in the pro gaggimate?

jniebuhr, 2025-03-12:
the Pressure Control will likely also use the 3WV if you program a sudden pressure drop. That wouldn't work in that case. Using a ramp where the pressure is only relieved through the puck should

closed 2025-03-19 by jniebuhr

dlking12, 2025-05-17:
Apologies for commenting on a close thread, however I’ve opened my machine to run through the install video and noticed the thermostats are completely different than those of other models and it looks as though it may not be compatible? Picture and link to the thermostats attached. 

Any help is appreciated!

https://www.mrbean2cup.co.uk/gaggia-steam-thermostat-l127-1nt-01l-0499

[image]