jniebuhr/gaggimate #141: [Feature Request] Purge after brew

DevNullGamer, 2025-03-15:
### Describe the feature

I would be nice to have a purge button on the post-brew screen.

This should then run a automated purge of the brewhead, like 2-5 seconds and maybe 1 or 2 times. (both variables could be configurable from the web ui).
This would allow people to easily purge some water to clean the showerscreen after a shot.

### Why is this feature useful?

Since the way you would do it currently, would be to start a new brew, and then use that as the purge mechanism. This way it would be a lot easier, and it would act as a reminder to clean the showerscreen post-shot.

### How complex is this feature?

Low - Simple enhancement

jonathanNL93, 2025-03-16:
Already requested by me :)

jniebuhr, 2025-03-16:
Duplicate of #121

closed 2025-03-16 by jniebuhr