jniebuhr/gaggimate #147: [Feature Request] Short Description of Feature

jonathanNL93, 2025-03-16:
### Describe the feature

Eureka Precisa Bluetooth scale support.
Sadly eureka precisa scales aren’t all compatible yet. 
They are quite random with their naming of the devices. 
Mine is quite old and doesn’t shows up when searching for scales. 

### Why is this feature useful?

Eureka precisa support 

### How complex is this feature?

Low - Simple enhancement

### Additional context

[image]
[image]

jonathanNL93, 2025-03-16:
I thought let’s make it an official issue then ;)

jniebuhr, 2025-03-19:
Is hopefully fixed by #158

jniebuhr, 2025-03-19:
Fixed in v1.4.2

closed 2025-03-19 by jniebuhr