jniebuhr/gaggimate #160: [Bug] Uncommanded exit from water mode while dispensing.

AndrewRussellHayes, 2025-03-25:
### Brief summary of the bug

Exit of water mode, into brew mode, while dispensing.

### Describe the issue in greater detail

I was dispensing water and when I looked up at the screen to press stop, it was on the brew screen ready to brew, but still actively dispensing water. I quickly flipped the power switch. I did not see anything else and it started back as expected. Sorry I don't have more information for you. I believe the version was 1.4.2. Thanks!

### Steps to reproduce

presumably you would press water. press dispense and then do nothing else. but i have not had it repeat.

### Are you able to reproduce the issue?

Yes

### Expected behavior

I would have expected the machine to remain on the water screen.

### Has this function/fetaure worked before with your GaggiMate?

Yes

### Relevant logs or error messages

```shell

```

### What machine are you experiencing this issue on?

GCP e24

### Additional context

Part of me wonders if I had hit update before this and it was hanging or something and then the display restarted into brew mode while the controller was still dispensing. I don't know if that is possible. I don't think I was updating but maybe I had hit it before and it stalled or something? I'm just reaching for ideas.

jniebuhr, 2025-03-26:
It is possible that the screen lost connection to the PCB. In this case the water should stop automatically after ~10 seconds. Was it dispensing water out of the steam wand?

AndrewRussellHayes, 2025-03-26:
yes it was dispensing water from the wand. additionally it happened again just now after an update. the change you mentioned on discord. was that in the version nightly: v1.4.2-2-g3b9fdcf ?

AndrewRussellHayes, 2025-03-26:
im wondering if there is a possibility that somehow the lego nature of my build has any impact here. i do not think it does but i should call out that its a lego build. should we add something to the template so people tell you what board/build type they have?

DevNullGamer, 2025-03-31:
We have seen a similar issue, and conducted a test today to verify an assumption on this.

This is related to the "done brewing" state, that wasnt cleared before moving to a new step (either because the machine was put into standby due to timeout or that the brew screen was exited without clearing the done screen). And this then makes the screen jump back to brew once it is cleared by pressing play on a different mode.

**Workaraound untill patched:**
Simply remember to click the checkmark on the "brew done" screen before moving to something else, avoid just swiping up to go directly to the menu or letting the machine timeout and go into standby in that uncleared state.

PR #168 (2025-04-01): fix: Fix jumping into brewscreen
Fixes #160
commit 1c6132ac (2025-04-01): fix: Fix jumping into brewscreen (#168)

Fixes #160
closed 2025-04-01 by jniebuhr