jniebuhr/gaggimate #169: [Feature Request] Adding EMC filter and fuse on line input

paulwunner, 2025-04-04:
### Describe the feature

Adding filtering and fuse in front of HLK-10M05 Power Module (line input)

### Why is this feature useful?

Although no EMC compliance is necessary, excessive noise on the line can disturb other devices in the vicinity.
Adding a fuse to reduce fire hazard.

### How complex is this feature?

Medium - Requires some changes

### Additional context

The datasheet of the HLK-10M05 has some information on the recommended filter and fuse.
Note that the claimed emc compliance would most likely, if at all, only be met with the mentioned filtering components.

jniebuhr, 2025-04-04:
The fuse was already added to the new boards. There is at least an EMC filter in the EU versions of the machines themselves.

paulwunner, 2025-04-04:
With a quick search I wasn't able to find anything about a filter. As the Gaggia Classic is a quiet "dumb" machine I also wouldn't really see why they would have added one. Can you give me a hint on where to find more about the said filter?

jniebuhr, 2025-04-04:
On the EU versions of the Gaggia Classic machines this board is installed:
https://theespressoshop.com/cdn/shop/files/5171.jpg?v=1724062053
There's no inductive filter coil on there but half of it is filtering capacitors and a varistor.
The other half is a capacitor based PSU with a relay to switch the machine one and off.

paulwunner, 2025-04-04:
Ah okay, well this looks like just differential mode filtering, which is also somewhat "far" away. Filtering would ideally be placed close to the noise source. Also for common mode filtering you would need the earth connection (a good one) on the board.
Anyway one has to see whether it's even a problem or not.

closed 2025-04-04 by paulwunner