jniebuhr/gaggimate #190: [Compatibility Request] Ascaso - Dream

spsancti, 2025-05-06:
### Producer of the machine

Ascaso

### Machine model

Dream

### Revision (if applicable)

2015

### Does the machine have a 3-way valve?

Yes

### What method of heating does your machine use?

Thermoblock

### Additional context

Single thermoblock machine which uses it for brew and steam. During steaming the pump is modulated to supply water to the heater similar to breville bambino

Fabrice31t, 2025-05-06:
I am also very interested. I would expect the internals of the Ascaso Dream 2015 and newer to be identical to the Ascaso - Steel Uno 2015 and newer.

spsancti, 2025-05-07:
Yes, Dream and Steel are basically identical inside up to wiring. 
There is older Dream and Steel Uno Prof that use boilers and are similar to Rancilio Silvia.

spsancti, 2025-06-02:
Okay, dreams with 3WV before 2019 without the draining tube are incompatible without replumbing, as the valve does not drain the group but only vents the thermoblock to the water tank. 
Dream UNO and PID, which have the draining tube, should be compatible similar to the Breville Bambino.