jniebuhr/gaggimate #191: [Compatibility Request] Ascaso - Steel Uno (without PID)

Fabrice31t, 2025-05-06:
### Producer of the machine

Ascaso

### Machine model

Steel Uno

### Revision (if applicable)

2015

### Does the machine have a 3-way valve?

Yes

### What method of heating does your machine use?

Thermoblock

Fabrice31t, 2025-05-06:
[Ascaso Dream-Arc-Basic.pdf](https://github.com/user-attachments/files/20072680/Ascaso.Dream-Arc-Basic.pdf)

[Ascaso Steel.pdf](https://github.com/user-attachments/files/20072700/Ascaso.Steel.pdf)

jniebuhr, 2025-11-05:
This machine is compatible, there's a bunch of build threads with more details on Discord. See this link: https://discord.com/channels/951416527721230336/1402686100895567955