jniebuhr/gaggimate #203: [Compatibility Request] Gaggia New Baby (single boiler)

ffegej, 2025-05-20:
> Hi! @jniebuhr  

Producer of the machine
Gaggia

Machine model
New Baby

Revision (if applicable)

Does the machine have a 3-way valve?
Yes

What method of heating does your machine use?
Single boiler

Additional context

> 
> How about the New Baby, a singe boiler machine similar to the Classic?
> 
> [image]
> 
> [image]
> 
>  Thanks!!!

 _Originally posted by @ffegej in [#112](https://github.com/jniebuhr/gaggimate/issues/112#issuecomment-2884756893)_

ffegej, 2025-05-29:
But it could be even greater!

0x64c, 2025-05-31:
I have completed a build with a New Baby 06; some things to note:
- may require drilling holes to bolt down the SSR
- heat sink may be a good idea for SSR if installed in bottom cavity of chassis
- if drip tray is allowed to over fill, water can fill up the bottom cavity of chassis, care should be taken to raise PCB and SSR above the bottom
- this machine comes in both hose/barb and ptfe/pressure fitting configurations
- the plumbing for the pressure sensor can be a tight fit, but it is possible! I mounted mine in the top near the boiler. hose/barb models will have more space to work with as the OPV is integrated into the pump instead of attached to boiler
- the controller board and SSR can be hidden in the cavity underneath the pump (in fact, there's nowhere else to mount it!)
- the casing does not have magnets to mount display, or any openings for the display power cable; front bezel can be removed (glued on from factory) and wire can be routed through some small openings, and screen casing screwed in to the plastic. alternatively, a new bezel can be machined (original is about 0.6~0.75mm thick) or 3d printed (perhaps integrating  tapered screws and a rear display housing?)
- display cable could alternatively be routed out the bottom (with added hole) and around the back or side, and screen mounted with VHB/foam tape/screws to the side or top of chassis

ffegej, 2025-06-28:
Thanks. This is very helpful.
I'll have to watch the install video and compare the diagrams. I'm a little nervous!
Do you have the Pro version? If so, have you tried the profiling?

0x64c, 2025-06-28:
Yep.

Also. I've come to realize that live and neutral are wired backwards compared to the GC/GCP models; you can see the difference by comparing the schematic diagrams; for the easiest install you may have to swap the two on the GM PCB and SSR.

ffegej, 2025-06-30:
Just to be sure, the model to get is the one for the Classic 1991-2009?

0x64c, 2025-06-30:
My point is, if you check the schematics, on the newbaby the switching circuitry is on the neutral side and the components are on the live side;  metal body gaggia of any year or generation seem to have it the opposite way.

So, newbaby is common live instead of common neutral.

ffegej, 2025-07-20:
Thanks again! I think you saved me from ordering the wrong kit. My unit seems to have hose/barb fittings. I presume that means I should get the Gaggia Classic Pro kit, right?

ffegej, 2025-09-17:
Hi all. I think this update is correct, but am always open to input.

[image]