jniebuhr/gaggimate #206: [Feature Request] Headless Mode

danielweber90, 2025-05-24:
### Describe the feature

I would like to request a "headless" mode option for GaggiMate, allowing the "Display" code to run on a generic ESP32-S3 (or similar ESP32 module) without requiring the LiliGo display hardware.
On first boot, the firmware could check for a compatible display. If no display is detected, the system should automatically fall back to "headless" mode.

### Why is this feature useful?

This feature would be highly beneficial for users who:

- Prefer a minimalist setup without an external display.
- Have limited space on their machine or enclosure for the display hardware.
- Seek a more affordable or flexible hardware configuration by using a standard ESP32 module they already have.

### How complex is this feature?

Low - Simple enhancement

danielweber90, 2025-06-30:
Yes, that was probably me.

Out of the box, the code won't run headless, as the `ui->init()` call in `src/display/core/Controller.cpp` will fail and cause the ESP to reboot.

However, it's already possible to run it headless with minimal changes: simply remove that call, and skip the screenReady check before connecting in `Controller::loop()`. You can see the required changes in [this commit on my fork](https://github.com/jniebuhr/gaggimate/commit/2ee8ab26e18fb049879827804c2124d90114f5a2).

Depending on whether the plan is to integrate this or keep it as a separate variant, it would be great to either:

a. Have headless mode enabled automatically if no display is found (this would likely require a proper implementation of `WaveshareDriver::isCompatible()`),
b. Or remove the UI library entirely to reduce binary size and make it work on smaller boards like the ESP32-S3-Zero.

In the absence of a display, the web UI becomes significantly more important. It would be great to expand its functionality — for example, by adding a shot timer.

jniebuhr, 2025-07-28:
Fixed by #335

closed 2025-07-28 by jniebuhr