jniebuhr/gaggimate #216: [Bug] Physical Mode Switch

spinasss, 2025-05-28:
### Brief summary of the bug

When turning Mode Switch back to brew After steam nothing happens.

### Describe the issue in greater detail

The physical button of mode Switch on my Gaggia Classic does instantly also trigger the pump Even if the temperature isnt achieved yet.

### Are you able to reproduce the issue?

Yes

### Steps to reproduce

1. Switch brew mode button in machine to steam
2. Switch back to brew

### Expected behavior

I think it should only activate the heating to steam temp. When switching Button back to brew mode, it stops the pump cycle (if its still active) or does nothing (if pump cycle has ended). I think it should go back into brew Mode then( start cooling down temp to brew)

### Has this function/fetaure worked before with your GaggiMate?

No

### Relevant logs or error messages

```shell

```

### Firmware version

1.4.4.

### What machine are you experiencing this issue on?

Gaggia Classic 2009

rich0664, 2025-05-28:
+1 on this, At the very least having steam button off go back to brew makes sense, but perhaps some level of customization to the button behaviour would be worth looking into.

031devstudios, 2025-05-29:
Yes I noticed the same. When you flip the Steam switch and changes setpoint and begins the Pump pulsing straight away. It should wait for setpoint then start pulsing automatically. At that stage you can open the steam valve.

jniebuhr, 2025-06-14:
Proposed redesign:
Switching on the steam switch or going to steam mode will not run the pump and show a "loading/heating" icon until the temperature is in +- 5°C range of the set temp. It will then start to actuate the pump. Switching off the steam button will then return to brew

jniebuhr, 2025-06-30:
Fixed by #277

closed 2025-06-30 by jniebuhr