jniebuhr/gaggimate #229: [Feature Request] Boiler Stabilized Indidicator

mconciatori, 2025-06-03:
### Describe the feature

Having some sort of indicator that the boiler is stabilized. 

### Why is this feature useful?

It would allow not relying on a person's memory or separate timer or gut feeling on when a shot is ready to be pulled.

### How complex is this feature?

Medium - Requires some changes

### Additional context

On a Gaggia Classic 10 minutes from cold is considered to be stabilized. Stabilizing after a shot can be 5 minutes, but that value is debatable.

Some implementations possibilities:
a) The feature could be time based only with required input from the user since it would be machine specific
b) Time based with a threshold around the setpoint. i.e. temp within setpoint +-1.5C for 5 consecutive seconds
c) Temp within setpoint +-1.5deg and heater command less than x percent for 5 consecutive seconds (this option could be machine independent, but the x percent could require some thought)
d) same as above, but the x percent could be learned on the users machine automatically. i.e. store the average heater command percent of that last 5 seconds before going to standby

mconciatori, 2025-07-04:
I recommend re-open this feature request. The current implementation seems only indicate when the temperature reading is stable which can happen soon after turning the machine on. At that point, the water in the boiler may not yet be stable and may not produce good coffee.

jniebuhr, 2025-07-04:
The feature was implemented the way you suggested. But I agree that it doesn’t necessarily mean the machine is saturated. The “correct” way to do this would probably be to measure the lag of the grouphead behind the boiler temperature on temperature changes and add that as a parameter. At this point we’d have many machine specific parameters and should probably offer machine presets