jniebuhr/gaggimate #248: [Compatibility Request] Breville/Sage - Barista Express

kojinism, 2025-06-15:
### Producer of the machine

Breville / Sage

### Machine model

Barista Express

### Does the machine have a 3-way valve?

Yes

### What method of heating does your machine use?

Thermocoil

### Additional context

Planning to purchase a new & better machine but Gaggia in my region costs double the price from US/UK. I'll probably look for other compatible models here in Github but since I already have a 5-year old Barista Express. I'd like to know if it's compatible so I can try.

scubafly, 2025-06-19:
I would buy the gaggimate instantly if it is compatible with my Barista express!

spsancti, 2025-07-07:
I installed GM into an old Ascaso Dream, which has a thermoblock. 
It works perfectly, with good temperature stability and steam better than stock. 
BBE has the similar capacity heater, pump and 3WV, so it should work, too. 
However, I assume that it will require full custom wiring because of all the smart features it already has.