jniebuhr/gaggimate #271: [Compatibility Request] Machine Name - Model

MattJR92, 2025-06-27:
### Producer of the machine

Rancilio

### Machine model

Silvia Pro X

### Does the machine have a 3-way valve?

Yes

### What method of heating does your machine use?

Dual boiler

### Additional context

I would like to use this for pre infusion and flow control on my Rancilio Pro X. Please verify if this is possible, thanks.

Viv9678tfg, 2025-10-05:
+

jniebuhr, 2025-11-05:
The Rancilio Silvia Pro X has been converted in various degrees. There's a couple of build threads on the Discord. The Gaggia Classic Pro kit should mostly work.
See this thread for reference: https://discord.com/channels/951416527721230336/1401083035666874410