jniebuhr/gaggimate #276: [Feature Request] Short Description of Feature

Sharpy-88, 2025-06-28:
### Describe the feature

Send a notification once device has reached stable temperature 

### Why is this feature useful?

Device typically takes around 15mins to reach a stable brew temperature, a means of alerting the user when it’s time to brew would be helpful 

### How complex is this feature?

Medium - Requires some changes

### Additional context

Not sure how notifications would work without a dedicated app but it may be possible to send an email as a starting point

jniebuhr, 2025-06-30:
You can do this right now with Home Assistant. We have added a stable indicator on the UI. Maybe there's also a way with HomeKit.

DevNullGamer, 2025-07-01:
You can with HomeAssistant, using the Notify method: https://www.home-assistant.io/integrations/notify/

Without home assistant, you would still need some 3rd party app to receive the notification, so home assistant would be the best approach to making such a service, in my opinion.

DevNullGamer, 2025-07-01:
Yes, to run HomeAssistant, you would need some kind of hardware.
But other then that, you would need some 3rd party software to send/receive notifications. You might be able to do something similar with Apple Home, if you have a hub for that alreday, but i dont so cant say for sure.

closed 2026-10-05 by jniebuhr