jniebuhr/gaggimate #296: Need help changing UI

dm6tt, 2025-07-10:
Hi guys,
I'm fiddeling around with Squareline and wanted to test the generated code on the display.
I just changed the png on the Init screen.
I followed [https://github.com/jniebuhr/gaggimate/issues/279] to generate to UI-code in the ui-directory.
When checking the generated code briefly, the new png and the code is beneath the ui folder with actual timestamp.
The code is then compiled and flashing works as well.
Afterwords the Init screen is still showing the old png.

I tried an erase_flash as well before flashing with no result.

What must be done additionally that the generated code is compiled and flashed?

Thanks for your support

Marcus

jniebuhr, 2025-07-17:
Did you correctly setup the path to go to `src/display/ui`?

jniebuhr, 2026-10-05:
/stale

closed 2026-10-05 by jniebuhr