jniebuhr/gaggimate #336: [Feature Request] Decrease temperature during phase of a shot.

jfox815, 2025-07-28:
### Describe the feature

I would like gaggimate to give me an option to drop the temperature during a phase of the shot. A simple on off of the heating element would work but also having a target temperature would be great. 

### Why is this feature useful?

A few of the popular profiles from Decent Espresso, such as the Low Contact and Ultra Low Contact, incorporate a significant temperature drop. I believe the simulated lever profile could also benefit from this feature, as many lever machines naturally experience a temperature drop toward the end of extraction.

### How complex is this feature?

Low - Simple enhancement

jniebuhr, 2025-07-28:
Already in #240

jniebuhr, 2025-07-28:
nothing to be sorry about, it's not exactly mentioned there. 😆