jniebuhr/gaggimate #374: [Feature Request] Bigger tap area for all actions

Sopamo, 2025-08-21:
### Describe the feature

First of all - thank you so much for this project! It was my first real electronics project and brings me a lot of joy.
I would like to contribute this feature, but want to ask first if it is something that would be accepted as a PR.

It happens to me regularly that I have to tap on start/back/increase temperature/decrease temperature multiple times and I'm almost 100% certain it's because the tap area of the actions is relatively small, so especially with me being quite tall and therefore the angle of my hand being a bit odd when using gaggimate it's not super intuitive for me to judge where I have to tap the button. I think there are probably many other users who at some point have mis-tapped. As there are only very few interactive elements on each screen I think there would really be no downside of making the tap areas quite a bit larger.

### Why is this feature useful?

Bigger tap areas = better UX :)

### How complex is this feature?

Low - Simple enhancement

Sopamo, 2025-08-22:
Amazing, excited to try it once it's released :)

jniebuhr, 2026-10-05:
Added

closed 2026-10-05 by jniebuhr