jniebuhr/gaggimate #389: [Bug] Gaggimate only connects Wifi Network when host name is changed

mikld86, 2025-08-27:
### Brief summary of the bug

My gaggimate v1.6 only connects to home wifi network when host name is changed from “gaggimate”

### Describe the issue in greater detail

After rebooting my wifi ap (unify AP) and router and changing the hostname  in the hostname field does my gaggimate connect to the wifi.
 

### Are you able to reproduce the issue?

Yes

### Steps to reproduce

1. Power machine on
2. Connect to Gaggimate Wifi AP
3. Enter to hostname as “gaggimate”
4. Hit Save and Restart button
5. Gaggimate will no longer connect to Home Wifi network. 
6. Restart Unify wifi AP, and Router
7. Gaggimate will still not connect. 
8. Login on Gaggimate Wifi AP
9. Change Hostname to “gaggi”
10. Click “Save and Restart”
11. Restart Wifi AP and Router
12. Gaggimate connects and is accessible via gaggi.local. 

### Expected behavior

Gaggimate is expected to connect to Wifi without changing default hostname. 

### Has this function/fetaure worked before with your GaggiMate?

Yes

### Relevant logs or error messages

```shell

```

### Firmware version

1.6

mikld86, 2025-08-31:
I think it’s due to how DCHP is handling reconnections. It needs to be flushed rather than being held onto. 

I’ve made a small code change (literally one line of code) in and issued a pull request so we’ll see if it’s implemented.

jniebuhr, 2026-10-05:
Fixed

closed 2026-10-05 by jniebuhr