jniebuhr/gaggimate #430: [Bug] Brew stops without reaching the target weight

meszinorbi, 2025-09-21:
### Brief summary of the bug

Machine stops brewing when reached time limit for a given phase even when a target weight is specified to stop at.

### Describe the issue in greater detail

With version 1.6.0 the brew stops when reaching the time limit of given phase without reaching the weight target.

I am using a Bookoo scale.

### Are you able to reproduce the issue?

Yes

### Steps to reproduce

1. power on
2. power on scale (wait for connection)
3. set brew type to go by weight
4. select a brew profile with "stop when" defined for weight reached
5. zero the scale
6. start brew

### Expected behavior

Expected would be to have the brew continue even after the set time is over if needed and only stop when given weight is reached.

### Has this function/fetaure worked before with your GaggiMate?

Yes

### Relevant logs or error messages

```shell

```

### Firmware version

both on 1.6.0

### Additional context

I thought the new software on the controller would be the issue so I downgraded that to 1.5.0 and kept the screen at 1.6.0 version, but the issue persisted. Only went away after downgrading both to 1.5.0.

jniebuhr, 2025-09-21:
Is this with a profile marked "pro"? We did change this behavior for pro profiles and kept it the same for the standard ones. 
On pro profiles a phase has a fixed duration and exits earlier on any stop condition. on the standard where you just have 2 inputs it switches between either.

meszinorbi, 2025-09-21:
It was with pro profile

meszinorbi, 2025-09-21:
There is not much to it than. It was an oversight from my side. Thank for the fast reply.

closed 2025-09-21 by meszinorbi